Zip-o-mat/Slic3r #2: I can't get it to work... Not sure what I am doing wrong

DarinDev1000, 2019-08-24:
### Version
1.3.0-dev

git describe --tag:  1.2.9-1367-g68877a0d

### Operating system type + version
LXQT Version: 0.13.0
lubuntu

### Behavior
I'm not sure If I am using it right. 
When I slice my stl, It does the normal layers on the top.
I'm not sure how to get the top layers to work. 

[image]
[image]
[image]

Any help would be appreciated!

#### STL/Config (.ZIP) where problem occurs
[STL_and_config.zip](https://github.com/Zip-o-mat/Slic3r/files/3536658/STL_and_config.zip)

Zip-o-mat, 2019-08-24:
There are two reasons this does not work.
First, the current implementation cannot handle scaling (This is a BUG). 
Workaround: Scale model in Slic3r, Right click->export as STL and then reopen the saved STL

Second: As the console then prints out, the necessary maximum print height is 10.4097 mm but your configuration has only a clearance of 8mm
Maximum print height is the difference between the lowest and the highest point of the nonplanar extrusion.

CAUTION this clearance has to be measured in the printer. If you configure a higher clearance than the printer can handle you risk a collision.

DarinDev1000, 2019-08-24:
Thank you! That fixed it!
Yeah, it was the scaling in slic3r. 
Apparently, the stl needs to be scaled correctly before import.
It does not work properly if the object is scaled in slic3r.
Thanks

closed 2019-08-24 by DarinDev1000