Zip-o-mat/Slic3r #3: [91%] Generating typemap failed

sebafudi, 2019-08-27:
### Operating system type + version
Ubuntu 18.04 LTS
Clean installation set up specifically for this build.

### Behavior
`make` comand at `[91%] Generating typemap`

```
Can't locate ExtUtils/Typemaps/Basic.pm in @INC (you may need to install the ExtUtils::Typemaps::Basic module) (@INC contains: ../local-lib/lib/perl5 /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.26.1 /usr/local/share/perl/5.26.1 /usr/lib/x86_64-linux-gnu/perl5/5.26 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.26 /usr/share/perl/5.26 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base).
BEGIN failed--compilation aborted.
xs/CMakeFiles/XS.dir/build.make:107: recipe for target 'xs/typemap' failed
make[2]: *** [xs/typemap] Error 2
CMakeFiles/Makefile2:150: recipe for target 'xs/CMakeFiles/XS.dir/all' failed
make[1]: *** [xs/CMakeFiles/XS.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

platsch, 2019-08-27:
Can't reproduce, but it looks like `ExtUtils::Typemaps::Basic` module is missing. 
Did you:

- export the perl library path correctly? `export PERL5LIB=<PATH_TO_SOURCE>/Slic3r/local-lib/lib/perl5` (replace PATH_TO_SOURCE with your actual path). Also try to export the path before executing `cmake ..` which would be the correct order anyway in my opinion...
- use the current HEAD of nonplanar branch? The `ExtUtils::Typemaps::Basic` dependency was just added by https://github.com/Zip-o-mat/Slic3r/pull/1 (the file is affected here: https://github.com/Zip-o-mat/Slic3r/pull/1/commits/4ffa4d39b4111292b9268597e6f3535804c40c74)

sebafudi, 2019-08-27:
I've used relative path to perl libraries. like this `export PERL5LIB=../local-lib/lib/perl5`
It built successfully when I've used absolute path like this `export PERL5LIB=/home/<USERNAME>/Slic3r/local-lib/lib/perl5`

Thanks!

closed 2019-08-27 by sebafudi