Zip-o-mat/Slic3r #4: No nonlinear layers

sebafudi, 2019-08-27:
### Version
#1 

### Operating system type + version
Ubuntu 18.04 LTS

### Behavior
Slic3r won't generate nonlinear layers. In console it says, that there's a collision. I've tried to use my printer and filament settings, but it was the same thing. Attached config is mostly default with only nonlinear settings changed.
[image]

#### STL/Config (.ZIP) where problem occurs
[stl_config.zip](https://github.com/Zip-o-mat/Slic3r/files/3545931/stl_config.zip)

platsch, 2019-08-27:
Yes, that probably is correct behavior, the model you are trying to slice has no surface which could be printed nonplanar without colliding with some other part of the model.
The collision checking is a tricky thing, it often finds collisions that are hard to spot for a human. That can probably be improved to allow for some tolerances or subdivide the object into areas which can still partly be printed nonplanar, but someone has to take the effort :-)
A visualization of collision areas would also be neat.

Zip-o-mat, 2019-09-04:
Maybe a better visualisation would help to see why there are no nonplanar surfaces generated.

ds-motivo, 2019-09-04:
I'm also _not_ getting any nonplanar layers, no matter my settings. Here's a ball (half of a pokeball) I got from thingiverse. I've also tried a few topological map models as well.

[image]

My nonplanar settings are 45 degrees nozzle and 25mm of clearance (just to test, not actually my printer settings).

ds-motivo, 2019-09-05:
OK I take back my previous post. I guess I'm not intuitively understanding the role of nozzle clearance height in this. Changing just a couple millimeters drastically affects your "reach" on certain geometries.

platsch, 2019-09-06:
Intuitively I would reckon 45° and 25mm should be sufficient to get at least some non-planar coverage.
My first guess for an explanation would be that slic3r only uses full facets and depending on the tesselation of your model it is possible that some facets span a higher distance than 25mm, preventing detection of a non-planar region.
Also: the error reports a collision...
@Zip-o-mat do you have an explanation?

Zip-o-mat, 2019-09-06:
First of all there is no existing printer(at least not that I know) with 45° and 25mm clearance.
There is a collision detected but it seems that this is not correct. I have to check the collision detection.
@ds-motivo can you upload the stl or post a link?

ds-motivo, 2019-09-06:
STLs attached below (in ZIPs). 

I only put the settings to 45deg/25mm because I figured it MUST produce some results. Couldn't make it happen on the pokeball, but _could_ make it happen on the cat toy.

I also have a file from a customer that I can't share that is a horn cover (the thing you press on your car's steering wheel), shallow top surface but couldn't get nonplanar layers with any settings.

[Cattoy1.zip](https://github.com/Zip-o-mat/Slic3r/files/3585794/Cattoy1.zip)

[pokeball_side.zip](https://github.com/Zip-o-mat/Slic3r/files/3585741/pokeball_side.zip)

[image]

wookie76, 2019-09-13:
Greetings,

I too was playing around with silly settings just to see how far it could go.  Is there a "soft" limit of 10mm clearance and 50°?  It didn't seem to change much beyond that.  It didn't seem to like straying too far from any of the default settings.

Thanks for the hard work,
Bryce

Zip-o-mat, 2019-09-16:
There should be no limits but at the moment the collision detection is rather sensitive and has many false positive collisions. I am working on the next version with better collision detection in nonplanar toolpaths.

wookie76, 2019-09-16:
That's great.  And again this is just nit-picking.  Overall it's quite
impressive what you've accomplished, thanks again.

ds-motivo, 2019-09-16:
Agreed! Keep at it, with enough traction I think this could influence the way 3d printers are designed. I'm already seeing a ton of the 3d printing youtubers take a crack at it.