Zip-o-mat/Slic3r #6: Builds for Windows

mbartlett21, 2019-09-01:
Is there anywhere were I can download a build of this for Windows?

Zip-o-mat, 2019-09-01:
sorry at the moment this is not possible. There will be  a precompiled version available soon, but I need some time.

jjpython, 2019-09-01:
How would one go about compiling this for Windows, I'm quite new to programming and compiling my own stuff, thanks in advance

platsch, 2019-09-03:
To build this on windows you could probably start here: https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-Windows

But: this repository is based on a branch which uses cmake instead of the default perl build script. That branch never made it upstream due to unfixable problems with perl under windows, see this PR for details: https://github.com/slic3r/Slic3r/pull/4168
So It's probably not straightforward...

Lorgie1984, 2019-09-05:
It is possible to get this working on Windows, especially on Windows 10 with enabled Linux Subsystem.

When you've enabled the Linux subsystem.

https://docs.microsoft.com/en-us/windows/wsl/install-win10

Than just follow the installation guide from here and install additionally Xming to get the GUI working:

https://www.howtogeek.com/261575/how-to-run-graphical-linux-desktop-applications-from-windows-10s-bash-shell/

And as you see, it works, a bit slow but it is okay.

[image]

Zip-o-mat, 2019-09-06:
Cool great!

OKUTAT, 2019-09-20:
I am very interested in the processing. My PC is Windows10. So I enabled "Windows Subsystem For Linux", installed Ubuntu and VcXsrv GUI tool, and built the Slic3r.

I managed to run the processing with a simple model like picture below. It was very fun. But it runs slowly. When I tried a little complicated processing, It stopped, and finally went down.

So, if using it in earnest, it would be better to use it on a normal Linux system. 

[image]

mjc619, 2020-06-08:
for people building on WSL now that 20 is out, follow this link to get a 18 or 19 LSB version in order to have all the libraries available for a smooth installation: https://wiki.ubuntu.com/WSL
I'd recommend the store version of [Bionic(18)LTS](https://aka.ms/wsl-ubuntu-1804) since it's probably the most common release for slic3r nonplanar on wsl.

efa, 2021-04-27:
can you please say if are you using WSL1 or WSL2 ?

andrewleppala, 2022-01-02:
> can you please say if you are using WSL1 or WSL2 ?

I can confirm this works on WSL2 with Ubuntu 20 (from the Windows store) on Windows 10 20H2. I followed @Lorgie1984 and their instructions, installing Xming on the Windows host and vim-gtk on WSL2. Framerate is pretty low and sometimes I have to adjust the nonplanar angle to another value and then back for it to properly slice something.

MikeySpring, 2022-01-20:
I've followed all the steps and it works up until the point where I try and start the gui. When I try the start command, I get the error, "Unable to init server: Could not connect: Connection refused 11:49:43: Error: Unable to initialize GTK+, is DISPLAY set properly?" I'm pretty new to coding, so maybe I'm misunderstanding and doing something wrong. Please let me know if you have any tips.

efa, 2022-01-20:
for DISPLAY related problems under WSL, you have to follow: https://github.com/microsoft/WSL/issues/4106#issuecomment-1000999401

MikeySpring, 2022-02-05:
> for DISPLAY related problems under WSL, you have to follow: [microsoft/WSL#4106 (comment)](https://github.com/microsoft/WSL/issues/4106#issuecomment-1000999401)
Do I have to do this step before or after I follow the Github instructions for the program?

MikeySpring, 2022-02-09:
Good news! I got the gui to work! @efa thank you so much for your link! Now I'm running into the problem of opening a file from my computer. None of the files I have from my computer show up. Does anybody know how to get passed this?

andrewleppala, 2022-02-09:
> Good news! I got the gui to work! @efa thank you so much for your link! Now I'm running into the problem of opening a file from my computer. None of the files I have from my computer show up. Does anybody know how to get passed this?

Your Windows drives are mounted in `/mnt/`

MikeySpring, 2022-02-24:
I'm trying to do some prints and keep running into collision problems. Does anyone have any rules of thumb for setting the Nonplaner Layer settings? I'm working on modifying my setup as well, but I want to get a better understanding of the Nonplanar Layer settings as well.

platsch, 2022-02-28:
> I'm trying to do some prints and keep running into collision problems. Does anyone have any rules of thumb for setting the Nonplaner Layer settings? I'm working on modifying my setup as well, but I want to get a better understanding of the Nonplanar Layer settings as well.

This is not related to windows build, please open a new issue.
Also provide more details, what exactly are you facing? Any screenshots?

mirakct, 2022-11-28:
Should anyone use mobaXterm as X11 server, I had to disable OpenGL acceleration to get it to work with the gui, otherwise it would throw a segmentation fault error. Cheers

penewold, 2023-03-17:
i get multiple errors trying to do the first step. It dosen't seem to be able to find the packages.

[image]

i have to using ubuntu 22, 20 and 18 (and used the right commands), but it still dosen't work.

please help. Thank you in advance

efa, 2023-03-17:
some time ago Debian switched the name of the package to 'libwxgtk3.0-gtk3-dev', and then Ubuntu follow as it is a derivative. Try that name

penewold, 2023-03-17:
i tried again on ubuntu 20 and it dosent give the error for that exact package but does still do many others.
[image]

penewold, 2023-03-17:
well thank you for your help which unfortunately was not succesful. I wish you the greatest kind sir. Goodbye

efa, 2023-03-17:
I think we need the help of this repo's author to update the code. Hope a rebase to the current Slic3r code, and finally a merge in the upstream repo

penewold, 2023-03-19:
I researched a bit and realised i forgot to use
`sudo apt-get update`
Which fixed all the problems i had. For anyone that had the same problem as me; try this.

efa, 2023-03-20:
it's a pity that PrusaSlicer: https://github.com/prusa3d/PrusaSlicer is not marked ad Slic3r: https://github.com/slic3r/Slic3r fork, while SuperSlicer: https://github.com/supermerill/SuperSlicer is a fork of PrusaSlicer