Zip-o-mat/Slic3r #7: unable to start with

neew1, 2019-09-03:
Ihello together, I'm new here and yes I have no idea of ​​ubuntu and meanwhile, but I am fascinated by the Odee nonplanar to print so much that I am already turning special nozzles.

I can not get it but with perl etc I can make these settings in slic3r. neither under windows nor under ubuntu someone could explain to me understandable which steps to make and in which order?

Thank you in advance

friendly greetings

mjc619, 2019-09-03:
There is a tutorial video available now for Windows users to make a VM and start running https://youtu.be/gmePlcU0TRw

neew1, 2019-09-03:
hey mjc619,
Thanks for the help, I've finally made it under ubuntu with cmake. the only thing I have not reached yet is a link so that I do not have to start with perl slic3r.pl every time. is this possible?

kind regards

platsch, 2019-09-03:
Your are looking for something like this https://help.ubuntu.com/community/UnityLaunchersAndDesktopFiles?

neew1, 2019-09-04:
I could not test it yet but that sounds exactly what I'm looking for.

Thank you so much

closed 2019-09-04 by neew1