Zip-o-mat/Slic3r #9: Non-planar bridging!

pediegoivan, 2019-09-04:
I think that the best feature that this non planar 3d print could provide besides surface quality is the ability to make bridges among non planar points! Saving a lot of material and time wasted in supports of oblique straight geometries.

pediegoivan, 2019-09-04:
Is this implemented? Can be added?

Zip-o-mat, 2019-09-04:
I don't know what you mean. But even with nonplanar layers you cannot print in thin air.

pediegoivan, 2019-09-04:
I mean. Bridging with a z hop

mjc619, 2019-09-04:
Building bridges while also moving z does not make them any better in surface quality than current bridging methods on one z layer. normal "bridging" is just a slicer's way of detecting that the layer below it was unsupported and will attempt to print with reduced flow and higher speed to prevent fails.

"nonplanar bridging" could be interpreted to be something like flying buttresses, where you have two heads of the bridge at different elevations and instead of printing many layers and support to connect the two. I don't know what you mean by z hop, (most) arches are not possible without support, even with nonplanar printing.
for a nonplanar bridge, this first of all would mean that 
- a nonplanar underside feature has already been put in place
- an effective support method that not only can even work with nonplanar undersides but be able to detect regions where nonplanar areas can somehow bridge themselves has been implemented
- a belt-driven z or delta config (the max speed that the z-axis can move prevents bridging speed bumps that can be possible with only x and y moves) is on the machine

bridging is a feature that was made to remove unnecessary supports if the printer has effective cooling and if a drop in bottom surface quality was acceptable. nonplanar cant even work with supports let alone be able to detect "bridgeable" undersides on a print. knowing even **how** to support nonplanar undersides is an enigma, external tree supports cant work, unless they are aware of the toolpath in order to support such moves, a boolean operation on the shadow of the mesh might work but would come with its own plethora of issues.

pediegoivan, 2019-09-04:
I meant. It would be cool to implement a bridging feature that could join two points with a different Z value in a straight line.
Bridging is an excellent way of reducing waste material. And a bridging feature supporting different Z positions during a bridging operation could enhance its applications.
Also, supports do not degrade the surface finish only when they are soluble. But, many 3d printers use a single material configuration making the use of detachable supports a must. In this sense, detachable supports do leave awful surface finish on pieces. Furthermore, bridging can give excellent results when propperly tuned and consumes less material and the risk of breaking a part trying to mechanically remove a support is neglected.

I know that this rectilitear bridging feature with two points at different Z values is not something that could change radically 3d printing. But i think that could be very useful.

Perhaps is not directly related to non-planar 3d printing... and this is not the right place to talk about this feature. I had this idea when I was looking at the non-planar 3d printing implementation. Probably that is why I made the connection with this and made this issue.

mjc619, 2019-09-04:
Most printers use screws on z, limiting bridge speed, it would cause more drooping than supported nonplanar moves and can't be done with chamfered or bevelled corners.

pediegoivan, 2019-09-04:
You are right. My only printer has a delta configuration and z the fastest axis, that is why I did not consider z speed.
I never paid attention to how ridiculously slow is the z screw mechanism of many printers. But maybe people interested in this feature could make it work. With very reachable 750 rpm and a 2mm pitch leadscrew, you get 25mm/s on the z-axis that would allow a 24.6° steep bridge at 60mm/s...
I do not know if it is worth though... I should test it with my delta.... but with manual Gcoding for tests.

platsch, 2019-09-06:
Okay. Trying to make an example. This could be a usecase for nonplanar bridging:
[image]

It would have to be printed in this order:
[image]

That could be possible if:

- 3 can be printed without collisions with the already printed part 2 (and other parts of the object if applied to anything more complex)
- 3 would have to be divided into a non-planar base layer and the filled with planar layers
- support could just be omitted below non-planar bridges

My impression without drilling too deep into it: while it might be possible, the logic is not exactly the same as for non-planar top layers. It's probably a lot of work for only a few models which could profit from the feature. I think the main issue is collisions, e.g. if region 2 and 3 are connected somewhere else in the object, it would not be possible to first print 2 and then connect to 3 with a bridge.

Zip-o-mat, 2019-09-06:
Ah ok now I understand what you mean. This seems to be a cool feature but this is a lot of work! If anybody want to implement this feel free.

AvinashPudale, 2019-09-10:
This can be done combination of modifying meshes and non planner printing. This going to awsome future. Hope get it lots of contribution