Zip-o-mat/Slic3r #11: OSX build

cronos45, 2019-09-08:
Is there Mac OS installation?

platsch, 2019-09-09:
No idea, never tested on an apple system. We don't even have one to run a test...
Your output doesn't look successful, what exactly have you tried to build it?

cronos45, 2019-09-15:
Used brew as needed to install, but can't get past perl Build.PL

galvesribeiro, 2019-09-20:
Hey folks!

Trying to build it on OSX as well but still have this issue:

```
$ cd build/
$ cmake ..
PATH: /Users/guto/dev/repos/Slic3r/local-lib/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/share/dotnet:~/.dotnet/tools:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/Applications/Xamarin Workbooks.app/Contents/SharedSupport/path-bin:/Users/guto/dev/tools/flutter/bin:/Users/guto/dev/tools/android-tools
PERL5LIB: /Users/guto/dev/repos/Slic3r/local-lib/lib/perl:
Slic3r will be built with GUI support
CMake Error at /usr/local/Cellar/cmake/3.15.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find PerlLibs (missing: PERL_INCLUDE_PATH) (found version
  "5.18.4")
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.15.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.15.2/share/cmake/Modules/FindPerlLibs.cmake:146 (find_package_handle_standard_args)
  xs/CMakeLists.txt:366 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/guto/dev/repos/Slic3r/build/CMakeFiles/CMakeOutput.log".

```

Any idea what that may be?

Thank you!

nyxaria, 2019-11-20:
try adding libextutils-cppguess-perl-native to dependencies