Zip-o-mat/Slic3r #13: Running Slic3r

xXxSpitfire187xXx, 2019-09-15:
Hi,

I did all as described, with the virtual machine and the windows app directly. I can not start anything in both cases.

> Can't locate Slic3r/XS.pm in @INC (you may need to install the Slic3r::XS module) (@INC contains: /home/test/Slic3r/local-lib/lib/perl5/5.26.1/x86_64-linux-gnu-thread-multi /home/test/Slic3r/local-lib/lib/perl5/5.26.1 /home/test/Slic3r/local-lib/lib/perl5/x86_64-linux-gnu-thread-multi /home/test/Slic3r/local-lib/lib/perl5/5.26.0/x86_64-linux-gnu-thread-multi /home/test/Slic3r/local-lib/lib/perl5/5.26.0 /home/test/Slic3r/local-lib/lib/perl5 /home/test/Slic3r/lib /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.26.1 /usr/local/share/perl/5.26.1 /usr/lib/x86_64-linux-gnu/perl5/5.26 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.26 /usr/share/perl/5.26 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base) at /home/test/Slic3r/lib/Slic3r.pm line 53.
BEGIN failed--compilation aborted at /home/test/Slic3r/lib/Slic3r.pm line 53.
Compilation failed in require at slic3r.pl line 16.
BEGIN failed--compilation aborted at slic3r.pl line 16.

In google I have found an answer, you should start the Build.PL again.

This did not work either

It is a step with the xs skipped

> App::cpanminus is up to date. (1.7044)
Class::XSAccessor is up to date. (1.19)
Devel::CheckLib is up to date. (1.13)
Encode is up to date. (3.01)
Encode::Locale is up to date. (1.05)
ExtUtils::CppGuess is up to date. (0.20)
ExtUtils::MakeMaker is up to date. (7.38)
ExtUtils::ParseXS is up to date. (3.35)
ExtUtils::Typemap::Basic is up to date. (1.05)
File::Basename is up to date. (2.85)
File::Spec is up to date. (3.75)
Getopt::Long is up to date. (2.51)
IO::Scalar is up to date. (2.111)
Module::Build::WithXSpp is up to date. (0.14)
Moo is up to date. (2.003004)
**skipping X/XS/XSAWYERX/perl-5.30.0.tar.gz**
Scalar::Util is up to date. (1.52)
Test::Harness is up to date. (3.42)
Test::More is up to date. (1.302168)
Thread::Queue is up to date. (3.13)
Thread::Semaphore is up to date. (2.13)
Time::HiRes is up to date. (1.9760)
Unicode::Normalize is up to date. (1.26)
threads is up to date. (2.21)
threads::shared is up to date. (1.59)

If someone could help me then I would be very grateful

Zip-o-mat, 2019-09-16:
The  skipping X/XS/XSAWYERX/perl-5.30.0.tar.gz seems normal. Have you tried to run the build again?
    mkdir build
    cd build
    cmake ..
    export PERL5LIB=<PATH_TO_SLIC3R_SOURCE>/local-lib/lib/perl5
    (e.g. /home/zip-o-mat/Slic3r/local-lib/lib/perl5)
    make

wookie76, 2019-09-16:
I was getting frustrated too, but realized I was just blindly copy-pasting too much without thinking.  
Remember:
<PATH_TO_SLIC3R_SOURCE>
is just a placeholder and you have to actually change it.
For me, running a terminal from the home screen, it was just /home/home/ but each instance may be different.

Zip-o-mat, 2019-09-16:
Great that this works now

closed 2019-09-16 by Zip-o-mat

xXxSpitfire187xXx, 2019-09-18:
hi,

thanks for the advices. i jused the folder of the source, maybe the wrong one. 

i will try

thank you