Zip-o-mat/Slic3r #14: Merge back the planar feature to the mainline

TQwan, 2019-09-28:
Hi,
I don't know if you were aware of this, but there is a pull request to merge back the planar feature to the mainline.
https://github.com/slic3r/Slic3r/pull/4864
They try to merge in your code to the mainline, but it was not accepted due to having too many conflicting changes.

I was wondering since if you might take a look, since most of them are your changes.

I think the solution would be something like: 
Create a new branch from the origin/master ( https://github.com/slic3r/Slic3r.git ) and then cherry pick your commits which only contains the actual planer slicing feature (also perhaps revert the space changes, to keep the diff readable).
Then squash it into a single commit and push it upstream to the new branch.
Finally create a new pull request from that branch.

Zip-o-mat, 2019-10-01:
Thanks for the info!
The code is not ready to be merged upstream at its current state

2403772980ygy, 2020-04-07:
I'm trying to force merge these branches to see if the conflicts do not matter when it comes to compiling. I'll report on this issue as soon as I got an advancement.