Zip-o-mat/Slic3r #15: Small STL files don't have non-planar sections even after scaling

karafso, 2019-10-13:
The STL I was trying to print was exported at a different scale than Slic3r uses. To fix that, I simply scaled it up by a large factor. However, all surfaces get ignored by the non-planar slicer because they are too small. Adjusting the scaling in Blender and exporting to STL, then loading it in Slic3r makes the non-planar parts work fine.

If possible, I think the non-planar slicer should take Slic3r's scaling factor into account.

Using version 1.2.9-1369-g01ec8b46

platsch, 2019-10-13:
Yepp, that's a known issue. Scaling is one of several things on the todo list.
As workaround you can just save the scaled model from slic3r (Object -> Export Object as STL) and reload the file.

karafso, 2019-10-13:
Thanks, that works. Should I close the issue?

platsch, 2019-10-13:
Unless @Zip-o-mat wants to keep it as a reminder, yes.

closed 2019-10-14 by Zip-o-mat