Zip-o-mat/Slic3r #16: Layers being reset to above part after generating gcode

pbcrazy, 2019-11-18:
Hi guys, great work so far! I am having an issue though, I am trying to use the nonplanar layers on an airfoil and it looks great....until I try to export the gcode. At that point, the nonplanar layers become planar above the part. Checking the gcode confirms this behavior, as the top 2 layers (the layers I want to be nonplanar) are all on the same Z height. Any idea what is causing this and how to fix it? Attached is the preview before generating gcode and after generating gcode.

Ubuntu version 18.04.3

[image]

[image]

Zip-o-mat, 2019-11-18:
I also had this behaviour. This usually happens when slice settings are changed when the object is already sliced. 
This is a Bug and needs to be fixed.
A workaround could be to set all settings and then slice load the object, slice and export gcode.

pbcrazy, 2019-11-18:
Quick slicing does indeed work, thanks for the tip! The issue didn't occur when settings were changed however. This is a very large assembly (I am also graduate student, working with a large-format printer) so maybe the code just isn't scaling well? I know Slic3r itself has problems with large layer heights.

Zip-o-mat, 2019-11-20:
Great that the slicing worked. It would be great to see some images of the final part. If you cant share it here, you can send me an email to ahlers@informatik.uni-hamburg.de.