Zip-o-mat/Slic3r #18: Issue running export line of code

maggiefannick, 2020-03-11:
I'm having issues with the export command not being able to export PERL5LIB. My code is below:

**maggie@maggie-VirtualBox:~$ cd Slic3r
maggie@maggie-VirtualBox:~/Slic3r$ git checkout nonplanar
M	Build.PL
Already on 'nonplanar'
Your branch is up to date with 'origin/nonplanar'.
maggie@maggie-VirtualBox:~/Slic3r$ cd build
maggie@maggie-VirtualBox:~/Slic3r/build$ cmake ..
PATH: /home/maggie/Slic3r/local-lib/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
PERL5LIB: /home/maggie/Slic3r/local-lib/lib/perl:
Slic3r will be built with GUI support
--  PERL_INCLUDE_PATH      = /usr/lib/x86_64-linux-gnu/perl/5.26/CORE
--  PERL_LIBRARY           = /usr/lib/x86_64-linux-gnu/libperl.so.5.26
--  PERL_EXECUTABLE        = /usr/bin/perl
--  PERL_SITESEARCH        = UNKNOWN
--  PERL_SITELIB           = /usr/local/share/perl/5.26.1
--  PERL_VENDORARCH        = /usr/lib/x86_64-linux-gnu/perl5/5.26
--  PERL_VENDORLIB         = /usr/share/perl5
--  PERL_ARCHLIB           = /usr/lib/x86_64-linux-gnu/perl/5.26
--  PERL_PRIVLIB           = /usr/share/perl/5.26
--  PERL_EXTRA_C_FLAGS     = -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include
--  PerlEmbed_ARCHNAME     = x86_64-linux-gnu-thread-multi
--  PerlEmbed_CCFLAGS      = -D_REENTRANT;-D_GNU_SOURCE;-DDEBIAN;-fwrapv;-fno-strict-aliasing;-pipe;-I/usr/local/include;-D_LARGEFILE_SOURCE;-D_FILE_OFFSET_BITS=64
--  PerlEmbed_CCCDLFLAGS   = -fPIC
--  LD                     = x86_64-linux-gnu-gcc
--  PerlEmbed_LDFLAGS      = -Wl,-E;-fstack-protector-strong;-L/usr/local/lib;-L/usr/lib/x86_64-linux-gnu/perl/5.26/CORE;-lperl;-ldl;-lm;-lpthread;-lc;-lcrypt
--  PerlEmbed_LDDLFLAGS    = -shared;-L/usr/local/lib;-fstack-protector-strong
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   log
--   locale
--   chrono
--   date_time
--   atomic
--   log_setup
--   regex
-- Configuring done
-- Generating done
-- Build files have been written to: /home/maggie/Slic3r/build
maggie@maggie-VirtualBox:~/Slic3r/build$ export PERL5LIB=/home/maggie/Slic3r/build
maggie@maggie-VirtualBox:~/Slic3r/build$ make
[  2%] Built target BSpline
[  8%] Built target poly2tri
[ 11%] Built target clipper
[ 72%] Built target libslic3r
[ 80%] Built target admesh
[ 83%] Built target libslic3r_gui
[ 85%] Built target ZipArchive
[ 87%] Built target polypartition
[ 90%] Built target nowide
[ 91%] Building CXX object xs/CMakeFiles/XS.dir/src/libslic3r/utils.cpp.o
[ 92%] Building CXX object xs/CMakeFiles/XS.dir/src/slic3r/GUI/wxPerlIface.cpp.o
[ 93%] Building CXX object xs/CMakeFiles/XS.dir/src/perlglue.cpp.o
[ 94%] Linking CXX shared library XS.so
CMakeFiles/XS.dir/XS.cpp.o: file not recognized: File truncated
collect2: error: ld returned 1 exit status
xs/CMakeFiles/XS.dir/build.make:243: recipe for target 'xs/XS.so' failed
make[2]: *** [xs/XS.so] Error 1
CMakeFiles/Makefile2:150: recipe for target 'xs/CMakeFiles/XS.dir/all' failed
make[1]: *** [xs/CMakeFiles/XS.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2**

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Zip-o-mat, 2020-03-12:
Try 
 export PERL5LIB=/home/maggie/Slic3r/local-lib/lib/perl5

maggiefannick, 2020-03-12:
I tried this several times before and got an error here every time as well. Here's the code:

maggie@maggie-VirtualBox:~$ cd Slic3r
maggie@maggie-VirtualBox:~/Slic3r$ cd build
maggie@maggie-VirtualBox:~/Slic3r/build$ cmake ..
PATH: /home/maggie/Slic3r/local-lib/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
PERL5LIB: /home/maggie/Slic3r/local-lib/lib/perl:
Slic3r will be built with GUI support
--  PERL_INCLUDE_PATH      = /usr/lib/x86_64-linux-gnu/perl/5.26/CORE
--  PERL_LIBRARY           = /usr/lib/x86_64-linux-gnu/libperl.so.5.26
--  PERL_EXECUTABLE        = /usr/bin/perl
--  PERL_SITESEARCH        = UNKNOWN
--  PERL_SITELIB           = /usr/local/share/perl/5.26.1
--  PERL_VENDORARCH        = /usr/lib/x86_64-linux-gnu/perl5/5.26
--  PERL_VENDORLIB         = /usr/share/perl5
--  PERL_ARCHLIB           = /usr/lib/x86_64-linux-gnu/perl/5.26
--  PERL_PRIVLIB           = /usr/share/perl/5.26
--  PERL_EXTRA_C_FLAGS     = -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include
--  PerlEmbed_ARCHNAME     = x86_64-linux-gnu-thread-multi
--  PerlEmbed_CCFLAGS      = -D_REENTRANT;-D_GNU_SOURCE;-DDEBIAN;-fwrapv;-fno-strict-aliasing;-pipe;-I/usr/local/include;-D_LARGEFILE_SOURCE;-D_FILE_OFFSET_BITS=64
--  PerlEmbed_CCCDLFLAGS   = -fPIC
--  LD                     = x86_64-linux-gnu-gcc
--  PerlEmbed_LDFLAGS      = -Wl,-E;-fstack-protector-strong;-L/usr/local/lib;-L/usr/lib/x86_64-linux-gnu/perl/5.26/CORE;-lperl;-ldl;-lm;-lpthread;-lc;-lcrypt
--  PerlEmbed_LDDLFLAGS    = -shared;-L/usr/local/lib;-fstack-protector-strong
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   log
--   locale
--   chrono
--   date_time
--   atomic
--   log_setup
--   regex
-- Configuring done
-- Generating done
-- Build files have been written to: /home/maggie/Slic3r/build
maggie@maggie-VirtualBox:~/Slic3r/build$ export PERL5LIB=/home/maggie/Slic3r/local-lib/lib/perl5
maggie@maggie-VirtualBox:~/Slic3r/build$ make
[  2%] Built target BSpline
[  8%] Built target poly2tri
[ 11%] Built target clipper
[ 72%] Built target libslic3r
[ 80%] Built target admesh
[ 83%] Built target libslic3r_gui
[ 85%] Built target ZipArchive
[ 87%] Built target polypartition
[ 90%] Built target nowide
[ 91%] Building CXX object xs/CMakeFiles/XS.dir/src/libslic3r/utils.cpp.o
[ 92%] Building CXX object xs/CMakeFiles/XS.dir/src/slic3r/GUI/wxPerlIface.cpp.o
[ 93%] Building CXX object xs/CMakeFiles/XS.dir/src/perlglue.cpp.o
[ 94%] Linking CXX shared library XS.so
CMakeFiles/XS.dir/XS.cpp.o: file not recognized: File truncated
collect2: error: ld returned 1 exit status
xs/CMakeFiles/XS.dir/build.make:243: recipe for target 'xs/XS.so' failed
make[2]: *** [xs/XS.so] Error 1
CMakeFiles/Makefile2:150: recipe for target 'xs/CMakeFiles/XS.dir/all' failed
make[1]: *** [xs/CMakeFiles/XS.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

Zip-o-mat, 2020-03-17:
Sorry for my late answer.
I cannot reproduce your problem.
Have you installed all dependencies, both 
 sudo apt-get install gcc cmake build-essential libgtk2.0-dev libwxgtk3.0-dev libwx-perl libmodule-build-perl git cpanminus libextutils-cppguess-perl libboost-all-dev libxmu-dev liblocal-lib-perl wx-common libopengl-perl libwx-glcanvas-perl libtbb-dev libxmu-dev freeglut3-dev libwxgtk-media3.0-dev libboost-thread-dev libboost-system-dev libboost-filesystem-dev libcurl4-openssl-dev libextutils-makemaker-cpanfile-perl
and
 sudo apt-get install libextutils-typemaps-default-perl libextutils-xspp-perl libextutils-typemap-perl
?

maggiefannick, 2020-03-17:
Yes, I've copied and pasted every line up to the point where I get the error.

Zip-o-mat, 2020-03-18:
Sorry but I can't help you because I cant reproduce the error and I have no idea how to solve this.
Maybe you can try a different Linux version I use Ubuntu 18.04 and it works there.

Zip-o-mat, 2020-03-18:
or you try my current work in progress branch:
´´´
$ git clone https://github.com/Zip-o-mat/Slic3r.git
$ cd Slic3r
$ git checkout -b origin/nonplanar-master-merge
´´´
And build with the following instructions:
´´´
$ export LDLOADLIBS=-lstdc++
$ perl Build.PL
$ perl Build.PL --gui
´´´

maggiefannick, 2020-03-19:
I am using Ubuntu 18.04 as well. Can't seem to get it to work. Thanks for the help anyway.

Markowia, 2020-04-22:
Hello @maggiefannick, 
I was on the same boat as you until now.
It was stopping and stopping again and again on 97% with the same error. Then I switched to WLS and did a try there. Surprisingly it was successfull and the Slic3r started but was cursed with bad performance. So I switched back to Ubuntu 18.04 virtual machine and replicaring the same steps i succeeded comppiling Slic3r (disclaimer: I am not a programmer and have very little experiance with Linux). This is what I did:
 
**Actual solution after the borring introduction:**

1. completely remove the folder /Slic3r, if you need - use `rm -rf dirname`
2 Open the terminal and before you begin all the steps overagain make yourself a super user permanently (or whatever it means) with: `sudo su`. Imput your user password and then...
3. repeat all the steps as described in the main page [https://github.com/Zip-o-mat/Slic3r/tree/nonplanar](url)

If this won't help you, then I'll be mind blown.

Stay safe and happy compiling :)