Zip-o-mat/Slic3r #20: Cant generate support while enable nonplanar slicing

wjk199511140034, 2020-03-30:
The nonplanar slicing is great work, I think very prospective! I buid this nonplanar branch in win10 with debian sub-system, its working fine at start, but today I found an issue though.
There is a fan model, and its should generate support like this
[image]

But, when I check both support and nonplanar, I get this:
[image]

Most worse is, it generate flat layer and support from 0 to Z max first, and fill nonplanar layer at last, that was definitely unprintable
And here is a model of z=sinx*siny, its a solidmodel, which means does not need support, there is nonplanar slicing result, pretty beautiful and perfect!
[image]

But if I check both support and nonplanar, boom...
[image]

platsch, 2020-03-30:
Yes, this is a known limitation, support generally doesn't work with nonplanar slicing at the moment and it is not trivial to fix this.
The problem is that support is computed from the polygons representing planar layers and nonplanar layers break with this assumption. Additionally, upport layers don't have the same height as normal layers, so the layer boundaries are not the same. Also: collision checking is an issue.
I think it is possible to find a solution for these problems, but it probably requires some research.
If anyone is willing to investigate this: feel free, would be really nice to have!

wjk199511140034, 2020-03-30:
> Yes, this is a known limitation, support generally doesn't work with nonplanar slicing at the moment and it is not trivial to fix this.
> The problem is that support is computed from the polygons [...]

I would like to investigate this further, but my programming level has limited me, and I still can't understand whole process of libslic3r. Maybe I can write a script to extract all the support generating lines from the gcode generated by normal print, and then write it to the gcode file which generated by just enabled nonplanar printing.

platsch, 2020-03-30:
I didn't think it through, but at least for some cases it could work. Slicer has support for post-processing scripts. However, such a script would certainly not be a general solution.

Zip-o-mat, 2020-03-30:
Hi,
Slic3r tries to support the nonplanar surface here, this produces these strange support generations. Pleas never try to print this, it will crash heavily and can damage the printer.

A workaround would be to deactivate support generation and use the option "Enforce support for the first X layers" and set the layer number just below the nonplanar surface. This will work as long as the nonplanar surface is the last surface of the model.

[image]

wjk199511140034, 2020-03-31:
> Hi,
> Slic3r tries to support the nonplanar surface here, this produces these strange support generations. Pleas never try to print this, it will crash heavily and can damage the printer.
>
> A [...]

Yes! its work, thank you very much !

pzemlyan, 2020-12-24:
The non-planar bottom layer of the model (that lay on support) will be also very important.

Ben-MSTR, 2021-01-25:
Hi,

I found the same kind of issue with support generation and sold it with support enforcement tip.
However, I do not succeed in creating non-planar bottom layers in addition with the top ones.
How could we manage this ?
Also, I may ask some stupid question but how about creating a first part with non-planar top layers and then print a second one over it ? This sounds tricky to me too but...

 By the way, thanks for this excellent 3D slicing work.

Zip-o-mat, 2021-01-26:
Nonplanar bottom layers are not implemented at the moment.

Ben-MSTR, 2021-01-26:
Thank you for your answer.
This confirms my thoughts but I was wondering since this comment :

> The non-planar bottom layer of the model (that lay on support) will be also very important.

And what about my naive proposal to consider support + bottom layers as a first print and core + top layers as a second print immediately above.
Thanks for your reply.