Zip-o-mat/Slic3r #23: Can't see the Non-planar parameter in Slic3r

Dethdok, 2020-05-23:
Hello everyone!

I'm totally new to github and programming language, but I wanted to try the Non-planar printing so i followed this github. I managed to follow all the steps explained in the README.txt, I can run Slic3r on the virtual machine. 

The problem is that I don't see neither the "Expert mode" in Slic3r nor the Non-planar settings. As i said before, I did all the steps in the ReadMe.txt without any error or major problem. 

I know this post is not professional, not sure if I need to give more information. Tank you so much in advance.

Zip-o-mat, 2020-05-23:
hi,

ist looks Like that you have compiled the wrong branch.

you need to Type:
   git checkout nonplanar

after you Clone the repository.

best,
daniel

Dethdok, 2020-05-23:
Hi Daniel,

I did the "git checkout nonplanar". It said im already on that branch. I tried changing the branch from nonplanar to master, and then change again to nonplanar. Opened Slic3r in both branches and nothing seems to change. 

Thank you so much for the fast reply.
Regards,
Adri