Zip-o-mat/Slic3r #24: Unable to Locate Packages

blackwings2114, 2020-05-23:
Hello, Trying to set it up, I have both a Ubuntu virtual machine and a Linux subsystem of tried on, but every time i try the code to download the necessary materials it comes up with packages not found, i made sure a firewall is not in the way. Any recommendations?

jin-eld, 2020-05-23:
I'm not using Ubuntu, but the same idea worked fine for me on Fedora - simply tell your package manager to install the build dependencies for the original slic3r, the deps should be more or less the same as for the nonplanar fork, i.e. something like: `apt-get build-dep slic3r` should get you going.

https://help.ubuntu.com/community/UpdatingADeb

Plantgrower, 2020-10-23:
While trying to install on Ubuntu 20.04 I had the following failures at the command:-  perl Build.PL

t/clean_polylines.t ............ 1/6 
#   Failed test 'Douglas-Peucker'
#   at t/clean_polylines.t line 20.
#     Structures begin differing at:
#          $got->[1][0] = '1'
#     $expected->[1][0] = '2'

#   Failed test 'Douglas-Peucker'
#   at t/clean_polylines.t line 28.
#     Structures begin differing at:
#          $got->[2][0] = '100'
#     $expected->[2][0] = '125'

#   Failed test 'gear was further simplified using Douglas-Peucker'
#   at t/clean_polylines.t line 68.
# Looks like you failed 3 tests of 6.
t/clean_polylines.t ............ Dubious, test returned 3 (wstat 768, 0x300)
Failed 3/6 subtests 
t/clipper.t .................... ok   
t/collinear.t .................. ok     
t/combineinfill.t .............. ok   
t/config.t ..................... ok   
t/cooling.t .................... ok     
t/custom_gcode.t ............... ok     
t/fill.t ....................... ok     
t/flow.t ....................... ok   
t/gaps.t ....................... ok   
t/gcode.t ...................... 1/27 
#   Failed test 'no Z moves below Z offset (no lift)'
#   at t/gcode.t line 70.
#          got: '130'
#     expected: '0'

#   Failed test 'no Z moves below Z offset (lift < z_offset)'
#   at t/gcode.t line 70.
#          got: '132'
#     expected: '0'
t/gcode.t ...................... 7/27 
#   Failed test 'no Z moves below Z offset (lift > z_offset)'
#   at t/gcode.t line 70.
#          got: '130'
#     expected: '0'

#   Failed test 'complete_objects generates the correct number of Z moves'
#   at t/gcode.t line 123.
#          got: '4148'
#     expected: '100'

#   Failed test 'complete_objects generates the correct Z moves'
#   at t/gcode.t line 124.
#     Structures begin differing at:
#          $got->[50] = Does not exist
#     $expected->[50] = '0.400'
t/gcode.t ...................... 18/27 
#   Failed test 'spiral vase is correctly disabled on layers with multiple loops'
#   at t/gcode.t line 216.
# Looks like you failed 6 tests of 27.
t/gcode.t ...................... Dubious, test returned 6 (wstat 1536, 0x600)
Failed 6/27 subtests 
t/geometry.t ................... ok     
t/layers.t ..................... 1/5 
#   Failed test 'wrong second layer height'
#   at t/layers.t line 38.

#   Failed test 'wrong layer height'
#   at t/layers.t line 42.

#   Failed test 'wrong second layer height'
[... 291 lines cut ...]
t/thin.t ....................... ok     
t/threads.t .................... ok   
t/vibrationlimit.t ............. ok   

Test Summary Report
-------------------
t/clean_polylines.t          (Wstat: 768 Tests: 6 Failed: 3)
  Failed tests:  1-2, 4
  Non-zero exit status: 3
t/gcode.t                    (Wstat: 1536 Tests: 27 Failed: 6)
  Failed tests:  5-7, 9-10, 23
  Non-zero exit status: 6
t/layers.t                   (Wstat: 1024 Tests: 9 Failed: 6)
  Failed tests:  3-4, 6-9
  Non-zero exit status: 4
  Parse errors: Bad plan.  You planned 5 tests but ran 9.
t/shells.t                   (Wstat: 768 Tests: 24 Failed: 3)
  Failed tests:  14, 16, 18
  Non-zero exit status: 3
t/skirt_brim.t               (Wstat: 256 Tests: 8 Failed: 1)
  Failed test:  2
  Non-zero exit status: 1
Files=35, Tests=594, 63 wallclock secs ( 0.11 usr  0.02 sys + 65.30 cusr  0.91 csys = 66.34 CPU)
Result: FAIL
Some tests failed. Please report the failure to the author!

closed 2021-05-25 by Zip-o-mat