Zip-o-mat/Slic3r #28: Segfault in mark_neighbor_surfaces()

coderforlife, 2020-09-18:
### Version
non-planar master version

### Operating system type + version
Fedora 32

### Behavior
Successfully compiled the program but the program either creates no non-planar layers (if the settings are too conservative) or crashes (segmentation fault) if I tweak the settings at all from the default of 10mm/10 degrees.

Using gdb I get the info that the segfault occurs at:
```
0x00007fffea6deaad in Slic3r::NonplanarSurface::mark_neighbor_surfaces(int) () from /home/user/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
```

At some point I will try to recompile with debug symbols so I can get a line number or something.

Additionally, if adaptive layers are used it crashes always (but maybe that is just because it is always finding non-planar layers)

[crashing.tar.gz](https://github.com/Zip-o-mat/Slic3r/files/5248570/crashing.tar.gz)

coderforlife, 2020-09-18:
I found out that it is a stack overflow due to the recursive nature of that function and it is hitting about 105,000 stack frames on my STL when it crashes. I was able to use ulimit to increase the stack size to 1 GB to allow it to get past `mark_neighbor_surfaces()` but I think that function should be re-written to not use recursion. It should be fairly straight-forward to do so and in the process become much more efficient likely. I will see if I can write it up later tonight or tomorrow using an `unordered_set<>`.

platsch, 2020-09-19:
The nonplanar feature currently relies on a fixed layer height. With adaptive layers it is unfortunately expected to fail or at least to produce weird results.
Anyway, great investigation. Looking forward to the pull request.

coderforlife, 2020-09-19:
Just tested and it worked so will make PR shortly.

With adaptive layers: the model I am working with has a base rectangle and then a complex surface above that. Can the custom layer heights be used so that the base (where there is no top surface) have one, thicker, layer height and anything near the top surface have a single, thinner, layer height?

Additionally, it seems to be not doing non-planar on much of the top surface of my model due to there being small, thin, peaks and it hitting the area cutoff. From my quick look through the thesis paper it seems this has to do with infill support under small areas or something. It is likely to fail if I lower the area limit in the code from 20 mm^2 to something like 1 or even 0.1 mm^2? Can I put a "ceiling" between the infill and the non-planar layer? Basically, my model seems to me to be perfect for non-planar printing (small single layer shifts, however the shifts don't last very long).

Zip-o-mat, 2020-09-22:
> Just tested and it worked so will make PR shortly.
Tanks for your code and sorry for my late reply.

> With adaptive layers: the model I am working with has a base rectangle and then a complex surface above that. Can the custom layer heights be used so that the base (where there is no top surface) have one, thicker, layer height and anything near the top surface have a single, thinner, layer height?
This should generally be possible when all layers that are in the range of the nonplanar surface have the same thickness. But test with caution!

> Additionally, it seems to be not doing non-planar on much of the top surface of my model due to there being small, thin, peaks and it hitting the area cutoff. From my quick look through the thesis [...]

The code removes a nonplanar surface and prints in planar when the surface area is below 20mm^2. This is just to not print very small areas as nonplanar when they only contain one or two very short lines. I agree that a hardcoded fixed value is bad coding style and should be changed but in general, this threshold is needed to prevent multiple very small nonplanar regions on complex objects that are messing around with the gcode.

Zip-o-mat, 2020-09-22:
#29

closed 2020-09-22 by Zip-o-mat