Zip-o-mat/Slic3r #34: Super Slicer - Non Planar

ETE-Design, 2020-11-28:
Would it be possible to add this as an Post Processing Script to Super Slicer? Or even Better just integrated it to the Slicer?

Zip-o-mat, 2020-11-30:
Since SuperSclicer is based on Prusaslicer which is based on Slic3r this should be possible.
But I don't have the time to do such a port.

ETE-Design, 2020-11-30:
@Zip-o-mat  Okey fair enough... Is it many files that need to be changes? Or is it possible to do by myself?

platsch, 2020-11-30:
Not so clear, I'm not aware how much the Prusa code has diverged by now.
One important thing is that most of the geometry base classes were mainly built for planar coordinates and had to be extended by the z-dimension (Point, Line). This currently is no clean implementation, somewhat inconsistent, and causes failures in several tests. Good enough for a research project, but maintainers are probably not so happy to merge it in the current state. Open issues with support structure etc. are another reason not to merge this into a production release.
Also: the GUI has changed from Perl based WX to C++ based WX. There are some tweaks required to visualize non-horizontal extrusions, that would have to be ported as well.

All in all: you are very welcome to spend some time on it, but you should be aware that is probably not done on a weekend and requires some knowledge of the slic3r codebase.

ETE-Design, 2020-11-30:
@platsch Thanks for the answer :-) Will there ba a Windows build then, instead of having to run Linux in a Box?

Zip-o-mat, 2020-12-01:
I have no idea how to build a windows version of the slic3r. If anyone knows please tell me how to do it. I tried building with the instructions here: [https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-Windows](https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-Windows)  but I can't get it to work.

Give it a try and tell me if it works.

ETE-Design, 2020-12-07:
@Zip-o-mat Mabye you could use the same way as Super Slicer compile for Windows?