Zip-o-mat/Slic3r #37: problem with running on WSL

akcent96, 2021-01-29:
I'm trying to run non-linear Slic3r on WSL Ubuntu 20.04 LTS on Windows 10 
I'm going straight forward with readme, and it does not work
while running build.pl everything goes forward, bu i keep getting that message every few minutes:

`/usr/include/boost/predef/detail/endian_compat.h:11:161: note: #pragma message: The use of BOOST_*_ENDIAN and BOOST_BYTE_ORDER is deprecated. Please include <boost/predef/other/endian.h> and use BOOST_ENDIAN_*_BYTE instead11 | RDER is deprecated. Please include <boost/predef/other/endian.h> and use BOOST_ENDIAN_*_BYTE instead")|                                                                                                      ^`

some of tests fails, and I get summary:

`Test Summary Report-------------------t/gcode.t                    (Wstat: 1536 Tests: 27 Failed: 6)Failed tests:  5-7, 9-10, 23Non-zero exit status: 6t/layers.t                   (Wstat: 1024 Tests: 9 Failed: 6)Failed tests:  3-4, 6-9Non-zero exit status: 4Parse errors: Bad plan.  You planned 5 tests but ran 9.t/shells.t                   (Wstat: 768 Tests: 24 Failed: 3)Failed tests:  14, 16, 18Non-zero exit status: 3t/skirt_brim.t               (Wstat: 256 Tests: 8 Failed: 1)Failed test:  2Non-zero exit status: 1Files=35, Tests=594, 134 wallclock secs ( 0.24 usr  0.14 sys + 135.00 cusr  2.33 csys = 137.71 CPU)Result: FAILSome tests failed. Please report the failure to the author!`

same with build.pl --GUI

and when i try to run slic3r.pl i get: 
`Unable to init server: Could not connect: Connection refused
22:22:19: Error: Unable to initialize GTK+, is DISPLAY set properly?
Failed to initialize wxWidgets at /home/akcent96/Slic3r/lib/Slic3r/GUI.pm line 7.
Compilation failed in require at /home/akcent96/Slic3r/lib/Slic3r/GUI.pm line 7.
BEGIN failed--compilation aborted at /home/akcent96/Slic3r/lib/Slic3r/GUI.pm line 7.
Compilation failed in require at (eval 85) line 1.`

tried all commands both with and without sudo, it does not help
I use VcXsrv for windows, Xming does not work either

No idea what to do next

magichappens89, 2021-02-17:
Have similar issues. The errors acutally start with:

```
Configuring /home/siebeckm/MyRepos/Slic3r/xs ... WARNING: the following files are missing in your kit:
        src/libslic3r/Zip/ZipArchive.cpp
        src/libslic3r/Zip/ZipArchive.hpp
Please inform the author.
```

So here I am informing you :).

Zip-o-mat, 2021-02-19:
Sorry I don't know why this version of slic3r does not work on WSL Ubuntu. Native Ubuntu works.

Regarding the test report: I know these tests fail, just ignore them.

I will take a look on the ZipArchive.hpp...

Zip-o-mat, 2021-02-19:
Ok I just fixed the boost issue and the ZipArchive issue. Please test if this works now

magichappens89, 2021-02-19:
Great thanks, I actually just got it running on 18.04. I just ignored the errors and it works! Haven't tried on 20.04 but it probably works as well. Thanks for the quick reply.

closed 2021-05-25 by Zip-o-mat