Zip-o-mat/Slic3r #39: Build on Ubuntu

Marco2929, 2021-03-13:
If I run the perl Build.PL I get this error
[image]

Bob90, 2021-03-13:
I got similar from lack of RAM allocated to VM

Zip-o-mat, 2021-03-15:
> If I run the perl command Build.PL I keep getting this error
> [image]

This should be fixed with commit 554cb82067ec7667abdf49b1a8d814fbe56df80d . Do you use the newest version? If not use git pull to get the recent changes.

IceChes, 2021-05-15:
I can confirm that this still is an issue after running "git pull" in the terminal in the correct directory.

Zip-o-mat, 2021-05-25:
You can try to delete everything and download it again. Also make sure that you are on the nonplanar branch.

closed 2024-04-03 by Marco2929