Zip-o-mat/Slic3r #42: undefined reference to `vtable for Slic3r::FillGyroid'

efa, 2021-04-10:
current git version: 1.2.9-1424-g08c8a8dd
LMDE 4 debbie, based on Debian 10 Buster

$ make
end at linking stage with this error:

```
[ 98%] Linking CXX executable slic3r
/usr/bin/ld: liblibslic3r.a(Fill.cpp.o): in function `Slic3r::FillGyroid::FillGyroid()':
Fill.cpp:(.text._ZN6Slic3r10FillGyroidC2Ev[_ZN6Slic3r10FillGyroidC5Ev]+0x1b): undefined reference to `vtable for Slic3r::FillGyroid'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/slic3r.dir/build.make:98: slic3r] Error 1
make[1]: *** [CMakeFiles/Makefile2:124: CMakeFiles/slic3r.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```
I googled around, but related search say to define the destructs and are already defined.

I figured out what is the link command from Cmake's Makefile and is:

`/usr/bin/c++   -std=c++11 -Wall -DM_PI=3.14159265358979323846 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DBOOST_ASIO_DISABLE_KQUEUE  -static-libgcc -static-libstdc++ -rdynamic CMakeFiles/slic3r.dir/slic3r.cpp.o  -o slic3r liblibslic3r.a libadmesh.a libBSpline.a libZip.a libclipper.a libexpat.a libpolypartition.a libpoly2tri.a -lboost_system -lboost_thread -lpthread -lboost_filesystem -lboost_chrono -lboost_date_time -lboost_atomic`

that seems right, so this is blocking for me.

### Add this side note:
at the end of '$ perl Build.PL' stage I got a lot of failed test with 'Please report the failure to the author!'

Note: _ before # is to avoid formatting as title
`
...
Successfully installed Slic3r-XS-0.01
1 distribution installed
t/adaptive_slicing.t ........... ok   
t/adaptive_width.t ............. ok     
t/angles.t ..................... ok     
t/arcs.t ....................... ok     
t/avoid_crossing_perimeters.t .. ok   
t/bridges.t .................... ok     
t/clean_polylines.t ............ ok   
t/clipper.t .................... ok   
t/collinear.t .................. ok     
t/combineinfill.t .............. ok   
t/config.t ..................... ok   
t/cooling.t .................... ok     
t/custom_gcode.t ............... ok     
t/fill.t ....................... ok     
t/flow.t ....................... ok   
t/gaps.t ....................... ok   
t/gcode.t ...................... 1/27 
_#   Failed test 'no Z moves below Z offset (no lift)'
_#   at t/gcode.t line 70.
_#          got: '130'
_#     expected: '0'
t/gcode.t ...................... 6/27 
_#   Failed test 'no Z moves below Z offset (lift < z_offset)'
_#   at t/gcode.t line 70.
_#          got: '132'
_#     expected: '0'

_#   Failed test 'no Z moves below Z offset (lift > z_offset)'
_#   at t/gcode.t line 70.
_#          got: '130'
_#     expected: '0'

_#   Failed test 'complete_objects generates the correct number of Z moves'
_#   at t/gcode.t line 123.
_#          got: '4104'
_#     expected: '100'

_#   Failed test 'complete_objects generates the correct Z moves'
_#   at t/gcode.t line 124.
[... 306 lines cut ...]
t/support.t .................... ok     
t/svg.t ........................ ok   
t/thin.t ....................... ok     
t/threads.t .................... ok   
t/vibrationlimit.t ............. ok   

Test Summary Report
-------------------
t/gcode.t                    (Wstat: 1536 Tests: 27 Failed: 6)
  Failed tests:  5-7, 9-10, 23
  Non-zero exit status: 6
t/layers.t                   (Wstat: 1024 Tests: 9 Failed: 6)
  Failed tests:  3-4, 6-9
  Non-zero exit status: 4
  Parse errors: Bad plan.  You planned 5 tests but ran 9.
t/shells.t                   (Wstat: 768 Tests: 24 Failed: 3)
  Failed tests:  14, 16, 18
  Non-zero exit status: 3
t/skirt_brim.t               (Wstat: 256 Tests: 8 Failed: 1)
  Failed test:  2
  Non-zero exit status: 1
Files=35, Tests=594, 77 wallclock secs ( 0.16 usr  0.04 sys + 79.81 cusr  1.20 csys = 81.21 CPU)
Result: FAIL
Some tests failed. Please report the failure to the author!
`

Zip-o-mat, 2021-04-24:
I never touched the FillGyroid. Maybe this is a general Slic3r issue or it is already fixed there.
Have you tried to build the standard slic3r?

efa, 2021-04-25:
OK, solved using last 'README.md' instructions and your hint in: https://github.com/Zip-o-mat/Slic3r/issues/43

closed 2021-05-25 by Zip-o-mat

tempodat, 2022-11-02:
Hey, would you mind sharing what you did to solve this? Ran into exact same problem with FillGyroid.

efa, 2022-11-02:
I solved using real Debian and not Debian inside WSL/Win10, see https://github.com/Zip-o-mat/Slic3r/issues/43