Zip-o-mat/Slic3r #43: Segmentation fault (core dumped)

efa, 2021-04-23:
### Version
Followed the new README.md build instruction.

nonplanar-master-merge branch
1.2.9-1408-g1b45df52

### Operating system type + version
gcc version 8.3.0 (Debian 8.3.0-6)
Debian 10 Buster

### Behavior
$ perl slic3r.pl
Segmentation fault (core dumped)

appear for a momento the GUI window, then crash.

Zip-o-mat, 2021-04-24:
Hi, please try the nonplanar branch

efa, 2021-04-25:
OK, with:
`$ git checkout nonplanar`
instead of:
`$ git checkout nonplanar-master-merge`
I can build and I got no segmentation fault.

Please update the 'README.md', users will follow the instructions there

efa, 2021-04-26:
work also with 'nonplanar-master-merge'.
The thing that not work is Debian inside WSL inside Win10, on a real Debian10 work with both branches.