Zip-o-mat/Slic3r #44: Slicing with non-planar and perimeters

LightningShark25, 2021-05-19:
### Version
1.3.0-dev zip-o-mat  (non planar)

### Operating system type + version
Ubuntu 20.04

I'm a reasonably experienced planar user, newbie to non-planar
Successfully compiled and spun up non-planar slicer and encouraged by the concept.

My model is a keyboard key

For the purposes of this excercise I've disabled via parameters the collision checking..  this is the result

[image]

This setting is with no perimeters.  As soon as I turn on vertical permiters, and 0 top layers it goes to this...

[image]

The non-planar part of this file now look good, but the top has lost its non-planar finish.
I can see that slic3r is trying to put a vertical permiter on the steps of the top, but really non-planar should take priority here?
Is this working as expected?
Is there any setting that controls this?

LightningShark25, 2021-05-19:
[key.txt](https://github.com/Zip-o-mat/Slic3r/files/6511889/key.txt)

Can't upload a .stl so I've renamed to txt - rename back to .stl to use

Zip-o-mat, 2021-05-25:
You need top solid layers because these are the layers that get projected as nonplanar.

LightningShark25, 2021-05-25:
Ah right.. I'll set top layers back to non zero and update with the result.

LightningShark25, 2021-05-25:
@Zip-o-mat I'm not super familiar with Git. Can you advise which branch is the latest? Assuming master but confused by master_merge

Zip-o-mat, 2021-05-26:
The latest branch is the nonplanar branch
```
git checkout nonplanar
```
I have just seen that the readme was still showing the old master_merge branch. It is now correctly referencing the nonplanar branch.

LightningShark25, 2021-05-29:
sorry for the linux/perl noob question.  I get the following error after trying to run perl slic3r.pl
I didn't get this on my last build... (admittedly using a different linux distro)

An easy fix?

Can't locate Slic3r/XS.pm in @INC (you may need to install the Slic3r::XS module) (@INC contains: /home/xubuntu/Slic3r/local-lib/lib/perl5/5.30.0/x86_64-linux-gnu-thread-multi /home/xubuntu/Slic3r/local-lib/lib/perl5/5.30.0 /home/xubuntu/Slic3r/local-lib/lib/perl5/x86_64-linux-gnu-thread-multi /home/xubuntu/Slic3r/local-lib/lib/perl5 /home/xubuntu/Slic3r/lib /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.30.0 /usr/local/share/perl/5.30.0 /usr/lib/x86_64-linux-gnu/perl5/5.30 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.30 /usr/share/perl/5.30 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base) at /home/xubuntu/Slic3r/lib/Slic3r.pm line 53.
BEGIN failed--compilation aborted at /home/xubuntu/Slic3r/lib/Slic3r.pm line 53.
Compilation failed in require at slic3r.pl line 16.
BEGIN failed--compilation aborted at slic3r.pl line 16.

LightningShark25, 2021-05-30:
So I redid the build and this time it works.... not sure what went wrong the first time as it seemed to complete the first time.  All good

Zip-o-mat, 2021-05-30:
Good to know. Happy printing

LightningShark25, 2021-05-31:
Ok so I have retested the latest non-planar branch.  It seemed that the hollow model with recessed text may have been causing extra complexity in the model so I simplified the example by using TeachingTech sample model.  Here is the output

[image]

As you can see, it has done something weird with the top layer and projected a solid square on top of the non-planar contour.

It may be a bug somewhere because if I then disable non-planar, then re-enable non-planar again, I get the following which looks more like what I'm after

[image]

I suspect it's something in the order of the settings selected that is causing this bug.  I had the same issue turning on and off 'thin wall detection'

If need be I can record a video reproducing the steps, but hopefully you can reproduce

LightningShark25, 2021-05-31:
PS  I saw this same behavior in the previous master_merge branch when I first built it.  This is why set perimeters to 0 which led to the first 'stepped' example I initially reported

LightningShark25, 2021-05-31:
[tt_5_non_planar_test.txt](https://github.com/Zip-o-mat/Slic3r/files/6569774/tt_5_non_planar_test.txt)

The STL file I used in the second example

Zip-o-mat, 2021-05-31:
yeah this is a known bug. 
The projection is not done correctly sometimes.
The workaround is to disable and enable nonplanar as you did.

LightningShark25, 2021-05-31:
Thanks for your time.  Happy to merge with the existing issue and close, or keep open till the bug is resolved....

Zip-o-mat, 2021-05-31:
Bug in #16

closed 2021-05-31 by Zip-o-mat