Zip-o-mat/Slic3r #48: XS will not install, everything else installs correctly

kittycatcaoimhe, 2021-11-01:
### Version
nonplanar

### Operating system type + version
openSUSE Tumbleweed 20211030 (kernel 5.14.14-1-default (64 bit))

### Behavior
XS will not install. When running `build.pl` abnormal output is:
`Configuring /home/*usr*/Slic3r/xs ... Slic3r requires the Boost libraries. Please make sure they are installed.
If they are installed, this script should be able to locate them in several
standard locations. If this is not the case, you might want to supply their
path through the BOOST_INCLUDEPATH and BOOST_LIBRARYPATH environment variables:
    BOOST_INCLUDEPATH=/usr/local/include BOOST_LIBRARYPATH=/usr/lib perl Build.PL
If you just compiled Boost in its source directory without installing it in the
system you can just provide the BOOST_DIR variable pointing to that directory.
N/A
! Configure failed for ./xs. See /home/*usr*/.cpanm/work/1635781629.26239/build.log for details.
The XS/C++ code failed to compile, aborting`
There is no file or directory named `boost` or `BOOST` other than the one in `~/Slic3r/src/xs/boost`
In dependencies installation, both `perl-threads` and `wxWidgets-ansi-devel` are not found. Used packages given for Fedora 30 installation as attempted replacement.
When trying to run `slic3r.pl`, output is:
`Can't locate Slic3r/XS.pm in @INC (you may need to install the Slic3r::XS module) (@INC contains: /home/*usr*/Slic3r/local-lib/lib/perl5/5.34.0/x86_64-linux-thread-multi /home/*usr*/Slic3r/local-lib/lib/perl5/5.34.0 /home/*usr*/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi /home/*usr*/Slic3r/local-lib/lib/perl5 /home/*usr*/Slic3r/lib /usr/lib/perl5/site_perl/5.34.0/x86_64-linux-thread-multi /usr/lib/perl5/site_perl/5.34.0 /usr/lib/perl5/vendor_perl/5.34.0/x86_64-linux-thread-multi /usr/lib/perl5/vendor_perl/5.34.0 /usr/lib/perl5/5.34.0/x86_64-linux-thread-multi /usr/lib/perl5/5.34.0 /usr/lib/perl5/site_perl) at /home/*usr*/Slic3r/lib/Slic3r.pm line 53.
BEGIN failed--compilation aborted at /home/*usr*/Slic3r/lib/Slic3r.pm line 53.
Compilation failed in require at slic3r.pl line 16.
BEGIN failed--compilation aborted at slic3r.pl line 16.`
Unsure if this issue has been reported before. When I looked, it seems the main Slic3r issues page has similar, but not the same issues, where the configuration of XS completes but the build fails. With mine, the configuration errors out.

_Is this a new feature request?_
No.

#### STL/Config (.ZIP) where problem occurs
N/A

Zip-o-mat, 2021-11-02:
It looks like the boost libraries are missing. Can you install it and retry?
Maybe this site helps: https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux

kittycatcaoimhe, 2021-11-02:
Which boost packages are necessary? I'm given 174 options when I do `zypper search boost`, and 20 of them are installed.

kittycatcaoimhe, 2021-11-02:
The only two that I'm finding people mentiom, `libboost-system-devel` and `libboost-thread-devel` are already installed

kittycatcaoimhe, 2021-11-08:
UPDATE: I have manually installed 80 boost packages through zypper, ran `perl Build.PL` with no change. I ran `sudo find /usr -name boost` and got:
`
/usr/share/doc/packages/perl-Boost-Geometry-Utils/src/boost
/usr/share/lmod/moduledeps/gnu-11/boost
/usr/share/lmod/moduledeps/gnu-11-mvapich2/boost
/usr/share/lmod/moduledeps/gnu-11-openmpi-4/boost
/usr/share/lmod/moduledeps/gnu-11-openmpi-3/boost
/usr/share/lmod/moduledeps/gnu-11-openmpi-2/boost
/usr/share/lmod/moduledeps/gnu-11-mpich/boost
/usr/lib/hpc/gnu11/mvapich2/boost
/usr/lib/hpc/gnu11/mvapich2/boost/1.77.0/lib64/python3.8/site-packages/boost
/usr/lib/hpc/gnu11/boost
/usr/lib/hpc/gnu11/openmpi4/boost
/usr/lib/hpc/gnu11/openmpi4/boost/1.77.0/lib64/python3.8/site-packages/boost
/usr/lib/hpc/gnu11/openmpi3/boost
/usr/lib/hpc/gnu11/openmpi3/boost/1.77.0/lib64/python3.8/site-packages/boost
/usr/lib/hpc/gnu11/openmpi2/boost
/usr/lib/hpc/gnu11/openmpi2/boost/1.77.0/lib64/python3.8/site-packages/boost
/usr/lib/hpc/gnu11/mpich/boost
/usr/lib/hpc/gnu11/mpich/boost/1.77.0/lib64/python3.8/site-packages/boost
/usr/include/boost
/usr/include/boost/chrono/typeof/boost
/usr/include/boost/hana/ext/boost
`
Running `find` with the path `/` provides only these, plus a bunch of the same within `/.snapshots` and `/home/*usr*/Slic3r/xs/src/boost`.
I ran `perl Build.PL` again with both `BOOST_INCLUDEPATH=/usr/include BOOST_LIBRARYPATH=/usr/lib/hpc/gnu11` and `BOOST_INCLUDEPATH=/usr/include/boost BOOST_LIBRARYPATH=/usr/lib/hpc/gnu11/boost` as well as trying `BOOST_DIR` for each, and it failed in the same way each time with the first two, and at the same spot with the message `Configuring /home/*usr*/Slic3r/xs ... Invalid BOOST_DIR: no such directory` with `BOOST_DIR`

Zip-o-mat, 2021-11-10:
Sorry I can not help you here. Simply because I also have no clue whats wrong.

isaac-abs, 2022-01-18:
I am not super familiar with any of this, but I was able to solve this error by increasing the memory I allotted to the virtual machine. I was running into this error when using 2GB, but was able to proceed past the XS installation failure at 5GB. Hopefully this helps.

kittycatcaoimhe, 2022-01-18:
> I am not super familiar with any of this, but I was able to solve this error by increasing the memory I allotted to the virtual machine. I was running into this error when using 2GB, but was able to proceed past the XS installation failure at 5GB. Hopefully this helps.

I am trying to compile natively on Linux with 64GB of RAM. I haven't tried in a little while though, so I will attempt to compile from source again tomorrow.