Zip-o-mat/Slic3r #52: Loading a *.stl File leads to  Sugmentation fault

ipat167, 2022-10-13:
### Version
1.3.1-dev

### Operating system type + version
Ubuntu 2204.1.6.0 WSL Windows 11

### Behavior
Loading a *.stl File leads to a Segmentation fault and Slic3r shuts down. I have tried different files but all give the same error.
The visualisation of Slic3r is done with Xmin.
[image]

Is it the Ubuntu version that is causing problems?

Zip-o-mat, 2022-10-17:
I can not reproduce the issue, because I don't use the WSL on Windows. I don't even have a Ubuntu 22.04 test system, maybe I will set one up in the near future.
I know that this is not a satisfying answer, but I can not help you at the moment.

ipat167, 2022-10-17:
I installed Ubuntu 18.04LTS WSL and it worked! I close the issue.

closed 2022-10-17 by ipat167