Zip-o-mat/Slic3r #54: GUI Issues

tacobotics, 2022-11-28:
Following the instructions, I tried installing this in a ubuntu vm. It does work, but when I try to open the gui I get this: 02:11:06 PM: Error: Unable to initialize GTK+, is DISPLAY set properly?
Failed to initialize wxWidgets at /root/Slic3r/lib/Slic3r/GUI.pm line 7.
Compilation failed in require at /root/Slic3r/lib/Slic3r/GUI.pm line 7.
BEGIN failed--compilation aborted at /root/Slic3r/lib/Slic3r/GUI.pm line 7.
Compilation failed in require at (eval 85) line 1.

Zip-o-mat, 2022-12-08:
It looks like your Wx is not properly installed.
What operating system are you using?
Did you follow the instructions from the [README.md](https://github.com/Zip-o-mat/Slic3r/blob/nonplanar/README.md) ?

tacobotics, 2022-12-08:
I have run all the commands in the README for the 20.08 version of Ubuntu.
I'm using Ubuntu 22.10 inside of a virtualbox vm in Windows 11. My Wx
seems to be good, [image: image.png]
Here is the error: [image: image.png]

It is quite possible that I missed something, as I am not very familiar
with slic3r. Here are the commands I've run, in the order that I ran them
in:

sudo apt-get install build-essential libgtk2.0-dev libwxgtk3.0-gtk3-dev
libwx-perl libmodule-build-perl git cpanminus libextutils-cppguess-perl
libboost-all-dev libxmu-dev liblocal-lib-perl wx-common libopengl-perl
libwx-glcanvas-perl libtbb-dev libxmu-dev freeglut3-dev
libwxgtk-media3.0-gtk3-dev libboost-thread-dev libboost-system-dev
libboost-filesystem-dev libcurl4-openssl-dev
libextutils-makemaker-cpanfile-perl

git clone https://github.com/Zip-o-mat/Slic3r.git

cd Slic3r

git checkout nonplanar

export LDLOADLIBS=-lstdc++

perl Build.PL

perl Build.PL --gui

perl slic3r.pl

perl slic3r.pl --gui

Thanks so much for the help!
Caleb

Zip-o-mat, 2022-12-12:
Hi, sorry for the late reply.

I can't see the images that you posted.
Is there a specific reason that you use 22.10? Otherwise, it is usually easier to get software running within the LTS versions of Ubuntu (16.04, 20.04  or 22.04)

tacobotics, 2022-12-12:
Ok, I used that version just because it was the most recent version. I'll
try those ones, and give an update

tacobotics, 2022-12-12:
Was able to get it running on 20.04, thanks so much for your help!

closed 2022-12-14 by Zip-o-mat