Zip-o-mat/Slic3r #56: Code Help

herfavor, 2025-03-18:
Hello, I have read your paper and i am very interested in it. I would like to reproduce the results and continue learning and researching based on it. However, I can only successfully compile the code repository, run the GUI, and slice the model. I am unsure how to find the part of the code where you modified Slic3r and how to make modifications. Could I consult you briefly on this? Also, do you have any further research results? I am very interested in non-planar printing. Thank you!

platsch, 2025-03-18:
So you are able to run the code (and it actually does nonplanar slicing) but now looking for good starting point to actually understand the modifications, correct?
I would recommend to start at [Print.hpp](https://github.com/Zip-o-mat/Slic3r/blob/nonplanar/xs/src/libslic3r/Print.hpp).
Just look for `nonplanar` in that file and from there find references to the methods elsewhere in the code.

herfavor, 2025-03-18:
Yes, thank you。

platsch, 2025-03-18:
Welcome. Looking forward to see interesting new ideas.